riscv/riscv-opcodes #14: Latex Control Sequence \instbit and \instbitrange not included

stephengeerlings, 2017-09-29:
I am missing these control sequences in the generated tex files. Am I missing a library? Or were these functions forgotten to be included?

aswaterman, 2017-09-29:
The generated latex files don't stand alone.  They are meant to be used in the context of https://github.com/riscv/riscv-isa-manual

stephengeerlings, 2017-09-29:
Thanks for the quick reply! Sorry for opening an issue!

closed 2017-09-29 by stephengeerlings

aswaterman, 2017-09-29:
That's what issues are for :)

pbreuer, 2023-07-10:
Can someone point to the .sty filein the archive where those macros are defined (better, note it in the table .tex file)? The table allegedly generated by pandoc from the priv-instr-table.tex file at  https://five-embeddev.com/riscv-isa-manual/latest/priv-insns.html#priv-instr-table seems to be absent without leave, and I would like to (re)create the missing table.

Regards

PTB

Fighter19, 2025-02-11:
@pbreuer 
The old commands were defined in this file:
https://github.com/riscv/riscv-isa-manual/blob/pause-public-review-20201013/src/preamble.tex