riscv/riscv-opcodes #27: Missing distinction between RV32 and RV64 (and probably also RV128 when the time comes)

tovine, 2019-05-09:
I was debugging some instruction generation, and came across an issue with the encoding of SLLI/SRLI/SRAI: In the ISA manual the 32-bit variant has a 5-bit `shamt` encoding, and the bits above (up until bit 30 in the case of SRAI) are required to be 0.
However the 64-bit variant uses 6-bit `shamt`, and because they are different base ISAs all the other encoding bits are the same as the 32-bit version.

This then causes problems when using the generated opcode mappings, as they allow for 6-bit `shamt` fields, while the 32-bit ISA forbids `shamt[5]` to be 1...

There may also be other instructions where this is a problem, is there a plan for tackling this?
It's a bit late to change the base ISA spec now, but one immediate solution (if it wasn't too late) would be to swap the encodings for SLLI/SRLI/SRAI and SLLIW/SRLIW/SRAIW in RV64I, which would then keep the instruction field boundaries concistent...

aswaterman, 2019-05-09:
There's already a way of handling this, just not a great one.  In opcodes-pseudo, there's `slli.rv32` etc. that do the right thing.  Whoever is consuming the output needs to know to use `slli.rv32` instead of `slli` when matching RV32 opcodes.

The same issue exists for RVC, where some opcodes have totally different meanings for RV32C and RV64C, and the same solution applies (in opcodes-rvc-pseudo).

tovine, 2019-05-10:
Got it, thanks for the tip!
Didn't think of looking for RV32 in there, would perhaps be nice with some mentions of it in the `README`

closed 2019-05-10 by tovine