riscv/riscv-opcodes #32: opcodes-rvc use of ignore considered unhelpful

tommythorn, 2019-08-04:
This may be debatable: the base `opcodes` table does a reasonable job a naming the individual fields, suchs as `rs1`, `imm12`, `bimm12lo`, `jimm20`, etc.  Enough that you can make a crude disassembler/assembler based on it as long as those fields names are treated correctly.

It seems that the treatment of compressed instructions is significantly different; instead of naming individual fields (and the myrid of immediate encodings) they have all been named `ignore`.

Wouldn't it be desirable to do the same for compressed as for uncompressed?  Eg. instead of
```
c.lw       1..0=0 15..13=2 12=ignore 11..2=ignore
```
we could have (untested)
```
c.lw       15..14=2 c.uimm6hi rs1' c.uimm6lo rd' 1..0=0
```
(along with the corresponding parse_opcode changes).

aswaterman, 2019-08-04:
IIRC, I used `ignore` out of laziness, given the plethora of RVC field types.  I don't see any technical disadvantage to your proposal.

If this were to be used by other tools, it might be useful to name distinct fields when the interpretations are different, even when they occupy the same bit positions.  For load instructions, for example, `c.lwimm` and `c.ldimm` could correspond to the same instruction bits, but they expand to different RVI immediates.

ghghost, 2022-01-26:
+1 to both tommythorn's comment (request) and aswaterman's proposal to distinguish fields with the same bit positions but different meanings. I wish I had the time to implement this improvement, but I don't.