riscv/riscv-opcodes #35: Missing RV32C shifts

tovine, 2019-09-09:
For the compressed shift instructions, `shamt[5]` is not allowed to be `1` per the spec, and the comment next to the listings say that `RV32 NSE, nzuimm[5]=1`.
Yet the opcode spec doesn't care about it: https://github.com/riscv/riscv-opcodes/blob/master/opcodes-rvc#L17

Therefore: should there not be RV32-specific versions of these instructions which declare this bit to `0`, like done for `slli.rv32` and similar?

aswaterman, 2019-09-09:
They exist: https://github.com/riscv/riscv-opcodes/blob/master/opcodes-rvc-pseudo#L11

tovine, 2019-09-09:
Ah, true. Sorry for not searching properly before posting, they didn't exist in my generated files and I didn't check all source files for updates

closed 2019-09-09 by tovine