riscv/riscv-opcodes #47: Why shamt in this repo is [25:20]

Phantom1003, 2020-08-19:
https://github.com/riscv/riscv-opcodes/blob/8d59eaae6d42e5dffa71846eb8296a8a5a461994/parse_opcodes#L36

Hi, I found that the opcodes in above line is [25:20], which should be [24:20] in spec. Is there is a miswrite, or I miss something ?

2henwei, 2020-08-19:
It means bit24-20 in Python

Phantom1003, 2020-08-19:
> It means bit24-20 in Python

No, I do not think so, rd is defined:
```python
arglut['rd'] = (11,7)
```
[11:7] follows description in spec
[image]

kito-cheng, 2020-08-19:
RV64 uses the bit 25 as the 6-th bit for shamt.

[image: image.png]

kito-cheng, 2020-08-19:
[image]

Phantom1003, 2020-08-19:
Got it, thanks, but I found that 32i also use the 6-bit shamt, may be there is a bug ?

aswaterman, 2020-08-19:
There's an alias for the RV32I version of SLLI: https://github.com/riscv/riscv-opcodes/blob/8d59eaae6d42e5dffa71846eb8296a8a5a461994/opcodes-pseudo#L2

Note that one uses `shamtw`, which uses bits 24:20.

closed 2020-08-20 by Phantom1003