riscv/riscv-opcodes #72: Missing 15. Vector Mask Instructions

drom, 2021-07-12:
The following instructions mentioned in spec:

## 15. Vector Mask Instructions

But missing in `opcodes-rvv` file

```
vmmv.m
vmclr.m
vmset.m
vmnot.m

vcpop.m
```

2henwei, 2021-07-12:
Those are pseudo-instructions. I don't think they should be handled here.

drom, 2021-07-12:
Agree, for
```
vmmv.m
vmclr.m
vmset.m
vmnot.m
```

Should they go to: `opcodes-rvv-pseudo` ?

drom, 2021-07-12:
As for `vcpop.m` it looks like rename:

`vpopc.m --> vcpop.m`

2henwei, 2021-07-12:
Ah I see. I didn't know there was pseudo functionality in the parser. Yeh, they could go to pseudo-files.

2henwei, 2021-07-12:
I just took a look through the parser. Seems that pseudo-instruction is not kinda same meaning as one in the assembler. It looks more like aliasing for instructions. I guess this is for legacy purpose like instructions being renamed and old names are retained.

PR #74 (2021-07-13): Updated several RVV instructions
Close #69, close #70, close #71, close #72, close #73

Some missing RVV instructions are pseudo-instructions for assembler to handle. I don't think they have to be added into the parser.
closed 2021-07-13 by aswaterman