riscv/riscv-opcodes #80: What is the meaning of this opcode?

nikita890-art, 2021-08-24:
mod rd rs1 rs2 31..25=1 14..12=0 6..2=0x1A 1..0=3

I am trying to understand what exactly this sentence means.
I found this line here,
https://nitish2112.github.io/post/adding-instruction-riscv/

what is the purpose of this line?
What is "14..12=0" , "6..2=0x1A" and "1..0=3" ?
What do they do?

2henwei, 2021-08-24:
They described values of these bitfields.

nikita890-art, 2021-08-24:
why is that they have values 0,0x1A and 3?
And how does that line depict mod operation?
@2henwei

aswaterman, 2021-08-24:
rtfc

2henwei, 2021-08-24:
@nikita890-art 
This line just describes the encoding of mod instruction, not the functionality.

nikita890-art, 2021-08-24:
oh, then suppose I want to develop an instruction that combines these 3 operations:
srli a5,s7,22
sili a5,a5,2
add s0,s0,a5
where do i describe the functionality?
and also, can I put any random values instead of 0,0x1A and 3?