riscv/riscv-opcodes #83: Spike crashing after adding new instruction!!

nikita890-art, 2021-08-24:
I created a new.h file in the riscv-isa-sim/riscv/insns folder to combine 2 instructions,

addi a5,gp,-1436 
add a5,a5,a3

So, the code I wrote to combine these instructions is,

int32_t v0 = sext_xlen(RS1 + insn.i_imm());
WRITE_RD(sext_xlen(v0 + RS1));

When I compiled and ran spike, it crashed.

I think there is something wrong with the logic of my code but I do not know how to fix it.

marcfedorow, 2021-09-10:
Neither of my custom instructions failed so I am sure it is not an issue of spike.
Consider doing this:
1) Ensure that your instruction is listed in encoding.h
2) Ensure that your instruction is listed in riscv/riscv.mk.in
3) Ensure there is no typos in match/mack/encoding/name/etc
4) Ensure that your instruction disassembles properly

chihminchao, 2021-09-13:
@nikita890-art 
Could you provide a branch pointing to your repo  or patches ? It is hard to tell what goes wrong.