riscv/riscv-opcodes #102: parse-opcodes script

takeBrian, 2022-02-13:
Kindly update parse-opcodes script, if possible. It tries to put encoding.h in the riscv-fesvr directory, whose repo is marked obsolete. Also, it seems encoding.h had been updated in other repos manually, leading to parse-opcodes breaking the previously working builds of the tools.

takeBrian, 2022-02-13:
Also, it should be mentioning python2 in the first line, instead of python.

aswaterman, 2022-02-14:
I think you might be using an old version of this repo.  All references to the `riscv-fesvr` repo were deleted in 2019.

takeBrian, 2022-02-14:
Okay. Actually, I didn't download anything from this repo directly, rather
i cloned "rocket-tools" and let it download submodules automatically, as
instructed on the "rocket-chip" GitHub page. I guess the ones it downloaded
automatically were older? Anyways, thanks for your prompt response.

takeBrian, 2022-02-14:
I figured it out: riscv-tools actually points to an older version of riscv-opcodes. I have to download the newer version of riscv-opcodes manually instead of the version downloaded automatically by the riscv-tools repo :) Though I've no idea whether it will work with other rocket-tools :( Oh well, I'll try, at least :)

aswaterman, 2022-02-14:
Oh, that makes sense.  Yeah, I can't speak to whether that'll work... rocket-tools is only meant to do the one thing it currently does.

closed 2022-02-14 by aswaterman