riscv/riscv-opcodes #111: rv32_i slli/srli/srai should not pseudo_op from rv64_i

ZenithalHourlyRate, 2022-05-03:
I'm trying to update rocket chip's `Instructions.scala` since the merge of #106. I found that `SLLI_RV32`, `SRLI_RV32`, `SRAI_RV32` are not generated. It turns out that `rv32_i` uses the opcode from `rv64_i` directly
https://github.com/riscv/riscv-opcodes/blob/833ba82b5df000ad6bc2a2cbca367d889272f88a/rv32_i#L1-L3

However, this would cause problem. In RV32 if the decoder is fed `slli rd, rs, 32` as encoded below, the result would be `slli rd, rs, 0` whereas originally this would cause an illegal instruction exception.
```
000000100000?????001?????0010011
``` 

Cc @neelgala

FYI
```
SLLI_RV32 = BitPat("b0000000??????????001?????0010011")
SLLI      = BitPat("b000000???????????001?????0010011")
```

neelgala, 2022-05-03:
@ZenithalHourlyRate even prior to #106 the `slli` instruction was considered a pseudo op of the RV64 `slli` instruction ([here](https://github.com/riscv/riscv-opcodes/blob/75b28f235e807fb6803575f952ce56b81571fd2b/opcodes-pseudo)). I have simply maintained that relationship here. 

Right now the parse.py script is not going to dump any pseudo-ops into the code if the original instruction is already present. This is because there are instructions like `slli` in RV32 which have the same mnemonics as the original instruction in RV64 but vary in encodings. This makes it difficult to create unique entries in the instruction yaml (even some code for that matter) for each instruction. 

If you want to build a decoder for rv32 you can use the following command : `make EXTENSIONS="rv32* unratified/rv32*"` which would generate the correct encoding for the use-case you have defined.

ZenithalHourlyRate, 2022-05-03:
> If you want to build a decoder for rv32 you can use the following command : `make EXTENSIONS="rv32* unratified/rv32*"` which would generate the correct encoding for the use-case you have defined.

Thanks for this information! Unfortunately rocket chip uses the same `inst.chisel` for both RV32 and RV64, so if I want to make rocket chip work I need to generate both file and merge them, which is painful and not maintainable.

> Right now the parse.py script is not going to dump any pseudo-ops into the code if the original instruction is already present.

Is it possible that the `parse.py` script could detect the difference in encoding and generate different entries when output encoding related codes. For mnemonics they could be the same.

neelgala, 2022-05-03:
> for both RV32 and RV64, so if I want to make rocket chip work I need to generate both file and merge them, which is painful and not maintainable.

If you are maintaining a single decoder for both RV32 and RV64 , then it would make sense to include the xlen checks in your decoder itself instead of overloading it in the inst.chisel. For example in rocket maybe moving [this check](https://github.com/chipsalliance/rocket-chip/blob/f544995f50d43cb2fe8e3c0883790095cb231c51/src/main/scala/rocket/RocketCore.scala#L182) inside [this class](https://github.com/chipsalliance/rocket-chip/blob/c3989b9284255a794f9e2f716e3ffa873952d2ba/src/main/scala/rocket/IDecode.scala#L198) would solve the problem ? This is similar to what [spike does](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/slli.h#L1) when encodings vary based on XLEN - it has further explicit encoding checks based on XLEN. 

> Is it possible that the parse.py script could detect the difference in encoding and generate different entries when output encoding related codes. For mnemonics they could be the same.

This is achievable but this would affect other artifacts as well - like the latex table generation. The instruction names in those tables will also change to `SLLI_RV32_I` which is not acceptable by the spec.

ZenithalHourlyRate, 2022-05-03:
> For example in rocket maybe moving [this check](https://github.com/chipsalliance/rocket-chip/blob/f544995f50d43cb2fe8e3c0883790095cb231c51/src/main/scala/rocket/RocketCore.scala#L182) inside [this [...]

Thanks! I will try to refactor that. Closing this issue now as it is deemed as an API change instead of a bug.

closed 2022-05-03 by ZenithalHourlyRate