riscv/riscv-opcodes #145: RV32_I has only 3 instructions?

CodePurble, 2022-11-05:
When using only the `rv32_i` `EXTENSION`, the generated masks are only for `slli`, `srai` and `srli` instructions. The rest are all CSR masks. Is this expected? Aren't there several other instructions in the base integer instruction set? The generated LaTeX table has several other instructions. What happened to all of those? Am I missing something?

CodePurble, 2022-11-05:
Okay, I just realized that I need to supply `rv_i` also to get those base instructions. I don't see this mentioned anywhere in the README. Here's the command if you need it:
`make EXTENSIONS='rv32_i rv_i'`

neelgala, 2022-11-05:
https://github.com/riscv/riscv-opcodes#file-naming-policy - this talks about the naming policy of the files

And also the Artifact Generation and Usage section has the following snippet:

> To select only a subset of extensions you can change the EXTENSIONS variable of the makefile to contains only the file names of interest. For example if you want only the I and M extensions you can do the following:

If you feel more information is required elsewhere please raise a PR and we can review it.

Thanks

CodePurble, 2022-11-05:
Ah the file naming policy covers it. Thank you so much for the information!

closed 2022-11-05 by CodePurble