riscv/riscv-opcodes #165: Constants.py is duplicated between riscv-opcodes and riscv-config?

tovine, 2023-02-27:
Seems very much like it is, should ideally be merged to a single source of truth and/or the one in this repository should get the same improvements as this? https://github.com/riscv-software-src/riscv-config/pull/105

neelgala, 2023-02-27:
They are not duplicated. I have used "constants" has a naming convention for a file holding manually defined constants. The constants.py file in opcodes and riscv-config is drastically different and used for completely different use-cases.

neelgala, 2023-02-27:
isa_regex in opcodes is probably useless.. I don't see any references to it and thus can be removed.
Thus, the RISC-V config version must be considered golden in this case

tovine, 2023-02-27:
Good. This of course needs to be investigated, as it might be used in various other repositories that use this as a submodule, but it sounds like a good idea to remove it if unused (and any such dependents should in that case be modified to use the one from riscv-config instead)