riscv/riscv-opcodes #171: Cannot turn tex files into PDF

gareth-anthony-hulse, 2023-04-01:
When attempting to turn `instr-table.tex` and `priv-instr-table.tex` into PDFs, it just spits out a ton of errors.  You can get the logs here: [latex-error-logs.zip](https://github.com/riscv/riscv-opcodes/files/11128505/latex-error-logs.zip) produced from [TeX Studio](https://archlinux.org/packages/community/x86_64/texstudio/).

**Specs:** https://linux-hardware.org/?probe=1146b0ebfc

neelgala, 2023-04-01:
those tex files are meant to replade the respective files in riscv-isa-manual repo and not meant to be used as standalone.

gareth-anthony-hulse, 2023-04-01:
I was hoping to get all the RISC-V instructions in a nice layout with binary representation, such as in the RISC-V spec from the chapter titled 'RV32/64G Instruction Set Listings'.

closed 2023-04-04 by gareth-anthony-hulse