riscv/riscv-opcodes #173: Use ULL (unsigned long long) suffixes in generated encoding.h file

JanMatCodasip, 2023-04-13:
Improvement suggestion -- note to self:

The integer literals in the generated `encoding.out.h` C header file should contain `ULL` suffixes (`unsigned long long`) when the actual value exceeds 64 bits.

This improvement has been requested in OpenOCD project which utilizes this generated .h file: https://review.openocd.org/c/openocd/+/7600

aswaterman, 2023-04-13:
This is a reasonable request, but whoever handles it needs to do it in such a way that the constants can still be used in assembly code, where the `ULL` suffix will cause a parse error.  See e.g. how the Linux kernel does this: https://github.com/torvalds/linux/blob/de4664485abbc0529b1eec44d0061bbfe58a28fb/include/uapi/linux/const.h#L7

JanMatCodasip, 2023-04-14:
> ... whoever handles it needs to do it in such a way that the constants can still be used in assembly code, where the ULL suffix will cause a parse error ...

@aswaterman, thank you for pointing this out and for the link. I will keep that in mind when making the change.