riscv/riscv-opcodes #175: encoding conflict between Zfbf/fcvt.bf16.s and Zfa/fround.h

aamartin0000, 2023-04-25:
here, in riscv-opcodes,
Zfa/fround.h           is encoded as    rd rs1 24..20=4 31..27=0x08 rm       26..25=2 6..2=0x14 1..0=3
Zfbfmin/fcvt.bf16.s is encoded as    rd rs1 24..20=8 31..27=0x08 rm       26..25=2 6..2=0x14 1..0=3

However, In the bfloat16 spec, 
Zfbfmin/fcvt.bf16.s is encoded as    rd rs1 24..20=4 31..27=0x08 rm       26..25=2 6..2=0x14 1..0=3
which is identical and conflicts with fround.h

For fcvt operations, 22..20 are used for the size of the source operand, but in riscv-opcodes [22] is set if the conversion is from/to bfloat16.  However, for zfa ops, [22] is an opcode bit for fround/froundnx.  Suggestion: change fround/froundnx to be [23:22]=2 instead of [23:22]=1.

aswaterman, 2023-04-25:
The Architecture Review Committee recommended to the BF16 TG to use the encoding used by here by riscv-opcodes.  I believe they agreed, but perhaps their spec hasn't been updated yet.

aamartin0000, 2023-04-25:
Not a BIG deal, but using [23] to signify bf16 in fcvt.bf16.s, but [22] to signify it in fcvt.s.bf16 is not as clean an encoding.

aswaterman, 2023-04-25:
`fround` etc. cannot be changed at this point.

The encoding oddity isn't a big deal, as you point out.  If you prefer, you can think of [23] as signifying "output operand has an alternate format" and [22] as signifying "input operand has an alternate format"...

aamartin0000, 2023-04-25:
sounds good!  Thanks for the quick response.

closed 2023-04-25 by aamartin0000