riscv/riscv-opcodes #180: Wrong encoding of Vector Mask-Register Logical Instruction

nrajovic, 2023-07-21:
Hi,

Seems like definitions of Vector Mask-Register Logical Instructions are not up to spec.
Namely, Chapter 15.1 of 1.1-draft Vector spec, says the following:

**Vector mask logical instructions are always unmasked, so there are no inactive
elements, and the encodings with vm=0 are reserved.**

However, lets take one instruction such as **vmand.mm**, its definition is as follows
(taken from rv_v file):

`vmand.mm       31..26=0x19 vm vs2 vs1 14..12=0x2 vd 6..0=0x57`

Here we see it says **vm** is a varible field, which is not in accordance with spec.

How do we proceed here, should I submit a fix for this? 

Thanks

aswaterman, 2023-07-21:
This is a holdover from an older version of the vector spec.  I'd appreciate it if you make a PR that fixes this.

nrajovic, 2023-07-26:
Fix is merged. Closing.

closed 2023-07-26 by nrajovic