riscv/riscv-opcodes #183: Wrong encoding of fmvp.d.x instruction

nrajovic, 2023-07-26:
Seems like **fmvp.d.x** is not in line with specification. Here is what spec states:

```
For RV32 only, if the D extension is implemented, the FMVP.D.X instruction moves a double-precision
number from a pair of integer registers into a floating-point register. Integer registers rs1 and rs2
supply bits 31:0 and 63:32, respectively; the result is written to floating-point register rd. FMVP.D.X is
encoded in the OP-FP major opcode with funct3=0 and funct7=1011001

```

The meaning of the above is that instruction consumes three registers, **rs1**, **rs2** and **rd** but this is how it is defined in **unratified/rv32_d_zfa**

`fmvp.d.x     rd rs1 24..20=1 31..27=0x16 14..12=0 26..25=1 6..2=0x14 1..0=3
`

From the above we see we define fmvp.d.x as consuming rd and rs1 while having rs2=1 which is not in line with spec.

aswaterman, 2023-07-26:
cc @ptomsich - looks like this must've been a copy-paste error because your Spike implementation correctly references rs2.

@nrajovic can you submit a PR that fixes this (and also `fmvp.q.x` in `unratified/rv64_q_zfa`, which appears to have the same error).

aswaterman, 2023-07-28:
@ptomsich Of course, I'm happy to merge whichever PR happens first :)

closed 2023-07-31 by ptomsich

aswaterman, 2023-08-03:
@nrajovic when you get the chance, can you update the `encoding.h` in Spike, so that this bug is fixed over there?

PR #184 (2023-07-31): fmvp.{q,d}.x: rs2 is variable field.
This resolves #183 .