riscv/riscv-opcodes #204: Unclear grammar / Exceptions to instruction syntax

tomhepworth, 2023-10-26:
I am having a go at writing a parser and I notice that most instructions are described by the format in the README  

IE `<instruction name> <instruction args> <bit-encodings>`

However there seem to be exceptions to this eg in `rv64_i`:

```
sraiw   rd rs1 31..25=32 shamtw 14..12=5 6..2=0x06 1..0=3
```

Is this a mistake or am I missing something

aswaterman, 2023-10-26:
The format is consistent; it's just that args and bit encodings may be intermixed.  The only requirement is that the name is the first token and the others come later.

closed 2023-10-26 by aswaterman

tomhepworth, 2023-10-26:
Perhaps the README should be updated to reflect this, since (at least to me) it currently implies that they are distinct. 

Perhaps just something like `<instruction name> <instruction args / bit encodings>`

tomhepworth, 2023-10-26:
Or better: `<instruction name> <arguments>` with a note that `<argument> := <bit encoding> | <variable argument>`

aswaterman, 2023-10-26:
I agree, and your most recent version is nicely concise while being accurate.  If you PR it, I'll merge it.

tomhepworth, 2023-10-26:
On it :)

sequencer, 2023-10-26:
FWIW, I wrote a Scala decoder here: https://github.com/sequencer/rvdecoderdb

tomhepworth, 2023-10-26:
Nice, I am making one in Haskell