riscv/riscv-opcodes #229: In encoding.h, MSTATEEN0_HENVCFGH is wrong

jhauser-us, 2024-02-19:
In "encoding.h", this line is wrong:

    #define MSTATEEN0_HENVCFGH 0x0100000000000000

A different bit in `mstateen0`, bit 62, which "encoding.h" calls `MSTATEEN0_HENVCFG`, controls access to all of `henvcfgh`, `henvcfg`, and `senvcfg`.  Register `henvcfgh` was added the same time as `henvcfg` and `senvcfg`, in Priv 1.12.

The bit position indicated by the line quoted above is for Priv 1.13, and controls access to `hedelegh`, which is new with Priv 1.13.

aswaterman, 2024-02-19:
Note also https://github.com/riscv/riscv-opcodes/blob/cd1623f46d585a3fc60b96e02acf2033c6dfa1a1/encoding.h#L178

aswaterman, 2024-02-19:
I think the right thing to do is to delete both `MSTATEEN0_HENVCFGH` and `MSTATEEN0H_HENVCFGH` macros and rewrite any dependent software to use the `MSTATEEN0_HENVCFG` macro.  This is a bit more disruptive, but it will ensure anyone depending on the old macros notices the change.

PR #230 (2024-02-19): Remove erroneous MSTATEEN0[H]_HENVCFGH macros
MSTATEEN0[H]_HENVCFG should be used instead.

Resolves #229
commit 31a4c2a5 (2024-02-19): Remove erroneous MSTATEEN0[H]_HENVCFGH macros (#230)

MSTATEEN0[H]_HENVCFG should be used instead.

Resolves #229
closed 2024-02-19 by aswaterman