riscv/riscv-opcodes #239: Update CLIC CSR_xINTSTATUS addresses, add CSR_xINTTHRESH

dansmathers, 2024-03-12:
update CSR_MINTSTATUS, CSR_SINTSTATUS, CSR_UINTSTATUS addresses to read-only csr addresses and add CSR_MINTTHRESH, CSR_SINTTHRESH, CSR_UINTTHRESH csrs.

spec: https://github.com/riscv/riscv-fast-interrupt

Please see pull #226

dansmathers, 2024-03-20:
pinging.  Can this be reviewed?  It holds up sail/arch tests/spike/gcc/llvm.   Thanks

sobuch, 2025-01-07:
Encoding header generated from this repo is also used in riscv-openocd, so having the correct register numbers here is desirable for targets implementing the CLIC spec (e.g. esp32c5).

Changes for latest spec are minimal (https://github.com/riscv/riscv-opcodes/compare/master...sobuch:riscv-opcodes:updated-clic-csrs), @dansmathers could you update your pull request?

Could someone review this?