riscv/riscv-opcodes #263: Include nf field in instruction name

moscickimilosz, 2024-07-18:
The nf field is treated as a variable arg within the rv_v instruction encoding file.

As I am interested in having all rvv instructions present in the YAML output, I was wondering whether a PR that expands this would get accepted? I believe this would align better with other extensions present in the dict e.g. Zimop and Zcmop.

I can see 3 possible solutions:
1. Expand rv_v file to include all of the instructions
2. Expand rv_v file to include all of the instructions, with the new instructions encoded as $pseudo_op (following what is done in rv_zimop and rv_zcmop)
3. Edit parse.py to expand dictionary only for the YAML output - avoid altering other outputs.

aswaterman, 2024-07-19:
Whatever we end up doing here, for backwards compatibility we need to make sure that the existing definitions that end up in `encoding.out.h`, `inst.chisel`, etc. are retained.

moscickimilosz, 2024-08-20:
> Whatever we end up doing here, for backwards compatibility we need to make sure that the existing definitions that end up in `encoding.out.h`, `inst.chisel`, etc. are retained.

Hi @aswaterman, 
Could you please take a look at the proposed solution in this [PR](https://github.com/riscv/riscv-opcodes/pull/274)?

Thanks

closed 2024-08-21 by moscickimilosz