riscv/riscv-opcodes #280: Improved software engineering quality suggestions

Timmmm, 2024-09-01:
Hi, I was wondering if you would be open to PRs to improve the general software engineering quality of this repo. This is the sort of thing I'm thinking of in increasing order of invasiveness:

1. Add `pre-commit`, tell it to format all the Python code using Black or Ruff and enforce in CI.
2. Add Pylint to `pre-commit`.
3. Restructure the code to turn it into an actual Python package with `pyproject.toml` etc. Publish it on Pypi.
4. Add static type hints.
5. Add Pyright (static type checker) to `pre-commit` to enforce them.

There's also some low hanging fruit in the actual code:

1. [Use argparse](https://github.com/riscv/riscv-opcodes/pull/185/files) ([Typer](https://typer.tiangolo.com/) is a better option but it's beneficial not to add a third party dependency.)
2. Switch from ad-hoc dictionaries to dataclasses.
3. Don't reparse the files 3 times.

I can do all that fairly easily but I thought I'd check if you would actually accept those improvements first. Seems like you might a bit review bottlenecked based on the open PRs (who isn't?).

aswaterman, 2024-09-02:
@neelgala is the author of the bulk of the Python code, so I'll defer to him.  But at first blush this seems reasonable.

IIITM-Jay, 2024-10-03:
Commenting here to have just a glimpse of the improvements in software engineering qualities such as Refactorization, Modularization and using as much Optimization techniques as we can for better maintainability and enhancing the readability as well. Tagging a PR also with reference to above. 

[Refactor and Optimization:: constants.py](https://github.com/riscv/riscv-opcodes/pull/264)

/CC @aswaterman and @rpsene

ethindp, 2025-11-21:
The use of other linters/type checkers/formatters other than ruff seems quite redundant. Is there a reason we don't just use ruff for everything? I'm pretty sure it already incorporates all the lints/type checking/formatting that the other major linters/type checkers/formatters do.

IIITM-Jay, 2025-11-22:
I somewhat agree with @Timmmm  at some places. But, to add on, Pylint has a robust plugin system, allowing users to write custom checkers to enforce highly specific, in-house rules. Ruff implements all its rules natively in Rust and does not currently support third-party plugins.
Additionally, Pylint still catches a category of deeper, semantic bugs as compared to Ruff currently.

But yeah, no doubt, Ruff does fast syntactic linting.

Timmmm, 2025-11-22:
Yeah probably worth still running Pylint. Unfortunately the only time I tried to add a lint to it (to detect accidental concatenation of strings) I found that it was impossible because its parser threw away that information. 🤷🏻‍♂️