riscv/riscv-opcodes #284: encoding.out.h contains a syntax error

retiredfeline, 2024-09-24:
Running make generates line 3022 with this:

#define INSN_FIELD_RS2=RS1 0x1f00000

which causes the C preprocessor complain that INSN_FIELD_RS2 is being redefined. How to fix this?

aswaterman, 2024-09-24:
@AFOliveira It looks like your pseudoinstruction changes introduced this bug.  Please file a PR that makes sure that this new pattern does not result in this new macro being emitted in `encoding.h`.

aswaterman, 2024-09-24:
@retiredfeline in the meantime, you can check out an older commit, e.g. 07b95c53a2585412f1a9044716445eb87af1e9d5

AFOliveira, 2024-09-25:
> @AFOliveira It looks like your pseudoinstruction changes introduced this bug.  Please file a PR that makes sure that this new pattern does not result in this new macro being emitted in `encoding.h`.

I'll address it as soon as I can, sorry for the flaw.

aswaterman, 2024-09-25:
No problem; thanks for offering to fix it soon.

aswaterman, 2024-09-30:
Resolved by #288

closed 2024-09-30 by aswaterman