riscv/riscv-opcodes #302: Acquire and Release operations

AFOliveira, 2024-10-22:
I noticed there are no acquire (.aq) , release (.rl) or acquire and release (.aqrl) operations in this repository. Is there any reason for this or can I also add them?

aswaterman, 2024-10-22:
In this repo they are represented as arguments rather than as part of the instruction mnemonic.  So maybe we don't need to do anything?  What problem is solved by adding the other variants?

AFOliveira, 2024-10-23:
Oh, I see them now in the arg lut. I hadn't noticed them earlier, and I think there are many reasons as to why they should be specified. I believe that they can definitely work as arguments and there I don't see a need for adding them. Thank you for the clarification and sorry for my miss here.

closed 2024-10-23 by AFOliveira