riscv/riscv-opcodes #308: Missing Opcode for cm.jt

JAYANTH-IITM, 2024-10-29:
Opcode is missing for the cm.jt , which is part of zcmt

JAYANTH-IITM, 2024-11-04:
@aswaterman can you please help here

aswaterman, 2024-11-04:
The issue is that they overlap each other in the encoding space.  If `c_index` is less than 32, then it is `cm.jt`, otherwise, it is `cm.jalt`.

JAYANTH-IITM, 2024-11-04:
yes but  in rv_zcmt file only " cm.jalt  c_index   1..0=2 15..13=5 12..10=0  is there . 
cm.jt , the instr itself is not there.

aswaterman, 2024-11-04:
How do you propose to solve the problem?  If you just add the instruction, you'll get an overlap error.  If you define it as a pseudoinstruction, it will execute correctly, but the description is technically incorrect (since it is not a pseudoinstruction).

IIITM-Jay, 2024-11-05:
@aswaterman To differentiate `cm.jt` and `cm.jalt `without overlap, if there is any unused bit or an available spare bit , especially in positions that aren’t strictly defined by the existing `cm.jalt` encoding, then I think that could be used as a _**"flag"**_ to differentiate `cm.jt` from` cm.jalt`.

But I don't think there is any such available bit to have the above solution to work out. Am I correct?

aswaterman, 2024-11-05:
@IIITM-Jay Correct, there is no such bit.  The logic is that if `c_index[7:5] != 0` then it's `cm.jalt`; otherwise it's `cm.jt`.