riscv/riscv-opcodes #321: Recommended Community Standards for the Repository

IIITM-Jay, 2024-11-14:
### Description
Referring to the [Community Standards being provided by the GitHub](https://github.com/riscv/riscv-opcodes/community) , would it be a better idea to add a **PR Template** and an **Issue Template**. Listing a few suggestive points so that a contributor or issue raiser would find useful to raise impactful and review-friendly PR(s) before actually raising it to the repository. The Issue Template would also help to go through the issues efficiently by a maintainer. 

### PR Template

- [ ] Description _(a short description explaining the intention behind raising it, would be better to keep it in few 1 or 2 lines)_
- [ ] Changes Made _(a numbered list of items comprising of changes made)_
- [ ] Before and After Comparison _(if exists)_
- [ ] Approach Used _(explaining why using that approach and the benefits of using it)_

### Issue Template
- [ ] Description _(a short para or listing them as item-wise would be better)_
- [ ] Missing Things
- [ ] Expected Results/ Outcome

IIITM-Jay, 2024-11-14:
@aswaterman let me know your view points. We can add templates so that a PR/ Issue raiser will get the editable templates before raising it by enabling the `markdown` files. They can change them as per their own wish, but the template will get render as soon as they raise it to have an idea about what this repository is expecting a raiser to provide.

Looping @rpsene if he is having some points to share regarding the same.

aswaterman, 2024-11-28:
Truth be told, I'm not a huge fan of these templates: many people ignore them, and so they end up adding clutter but little value.  But if the other users of this repo have a favorable opinion, I'm happy to go along.

aswaterman, 2026-01-07:
Thanks for opining.  I'll close this, at least for now.