riscv/riscv-opcodes #342: Zimm vs Zimm5

AFOliveira, 2025-03-12:
Is there any difference between Zimm vs Zimm5? I have looked intro the instructions they are used in, e.g.

zimm:
`csrrwi    rd csr zimm       14..12=5 6..2=0x1C 1..0=3
`
zimm5:
`vaeskf1.vi   31..26=0x22 25=1 vs2 zimm5      14..12=0x2 vd 6..0=0x77
`

and even looked the spec for them, from what I can understand, they are exactly the same - an unsigned immediate. Is my assumption right or am I missing something?

aswaterman, 2025-03-15:
AFAICS they are the same.

`zimm` dates from when it was the only zero-extended immediate.  Since there are multiple sizes of zero-extended immediates now, `zimm5` is probably a better choice if we decide to standardize on one.

AFOliveira, 2025-03-17:
May I PR the change to have the same immediate on all variables?

aswaterman, 2025-03-17:
Fine with me.

PR #347 (2025-03-18): Change zimm variable to zimm5, since they were exactly the same. 
Fixes #342.
commit 9bd0acd7 (2025-03-18): Change zimm variable to zimm5, since they were exactly the same. Fixes #342. (#347)
closed 2025-03-18 by aswaterman