riscv/riscv-opcodes #344: build failure

ved-rivos, 2025-03-15:
I encounter this failure. Please help.
```
ved:~/rvi-code-forks/riscv-opcodes$ make
Traceback (most recent call last):
  File "./parse.py", line 25, in <module>
    extensions: list[str],
TypeError: 'type' object is not subscriptable
make: *** [Makefile:18: everything] Error 1

```

aswaterman, 2025-03-17:
You need to use python 3.9.  It's not great that we require 3.9 because of one line of code, but I didn't push back because 3.9 has been out for 4.5 years at this point.

closed 2025-03-17 by ved-rivos

ved-rivos, 2025-03-22:
Leaving this here - using a newer python alongside the system python - in case it helps someone:

```
curl https://pyenv.run | bash
echo 'export PYENV_ROOT="$HOME/.pyenv"' >> ~/.bashrc
echo 'command -v pyenv >/dev/null || export PATH="$PYENV_ROOT/bin:$PATH"' >> ~/.bashrc
echo 'eval "$(pyenv init -)"' >> ~/.bashrc
pyenv install 3.9.17
cd projects/myproject
pyenv local 3.9.17
python -m venv .venv
source .venv/bin/activate
```