riscv/riscv-opcodes #351: Some fcvt instructions do not need rm

MutualStyle, 2025-04-14:
Some of the fcvt instructions (which convert a low-precision number into a high-precision one) always produce exact value and is unaffected by rounding mode. 
And the riscv-gnu-toolchain assembler regards these fcvt instructions with rounding mode illegal. https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1522
Should rm be removed from the variable field of these fcvt instrucions?

pdonahue-ventana, 2025-04-14:
That was a toolchain regression at some point.  We have tests that specify all rounding modes for these instructions, specifically to make sure that the rm field has no effect.  The toolchain started refusing to assemble those tests at some point.  The ISA says that there's a field here and the field is optional, so it's not harming people who don't specify it.  I don't understand why the toolchain would suddenly refuse to allow you to specify it.  Just because a field is not useful doesn't mean that the toolchain should force you not to use it.

aswaterman, 2025-04-14:
Right, these _are_ fields in the instruction, even though they aren't useful for software.  DV teams still want to make sure they perform their stated function.

FWIW, I can understand why binutils would make a different decision from riscv-opcodes, but in the riscv-opcodes context, we want these fields to remain present.

closed 2025-07-18 by aswaterman